PiInTheSky/pits #3: Video stream save to SD-Card?

xpix, 2015-03-25:
Wonderful project, dave if it possible to add the feature to save the video stream to SD-Card during the fly and send also still images via SSDV?

daveake, 2015-03-25:
Photography is controlled by the "camera" shell script.  You could very 
easily edit this to take a short video, then a still image for the SSDV, 
then repeat.

The GPU supports creating a jpeg "on the fly" from the video every few 
seconds, say, but I don't know if there's software support for it yet.  
Take a look at the latest raspivid program to see if it does.

We don't plan to add the above to the PITS software ourselves.

Dave

xpix, 2015-03-25:
Thanks for ur fast answer, ok another solution. Make 1minute videos and extract a picture via ffmpeg for SSDV. Here an example:

  ffmpeg  -i video.h264   -y capture%d.jpg

I want to try this :)

closed 2015-04-11 by daveake