PiInTheSky/pits #9: Abusive path for APRS

rlharris, 2015-10-27:
Any path (e.g WIDE1-1, WIDE2-1) for a HAB is frowned on in North America since it adds congestion to the network while the balloon is aloft but such a path is valuable for recovery operations when the equipment is near the ground.

It appears that it is a simple change to make the ax25 frame contents change with altitude. I have modified aprs.c to switch between no path and the WIDE1-1, WIDE2-2 at 6000 feet. The value of 6000 feet was selected somewhat arbitrarily given the height of the White Mountains in New Hampshire (the area over which it is anticipated that the balloon will be making its descent). I hope the students with whom I am working will be able to test it.

rlharris, 2015-10-28:
After talking to a commercial pilot that has flown with a 300mW APRS transmitter all over North America, it appears that 10,000 feet would be a good transition from no path to the WIDE1-1, WIDE2-1. He even suggested dropping the WIDE1-1 completely

Upuaut, 2015-10-28:
Hi there,

There is no "official" guidance on this and its basically down to the operators discretion. When doing the PITS code we followed these guidelines :

http://www.arhab.org/aprs

However as you raised the issue we will look at making this configurable by the operator possibly even with altitude  settings as you have requested.

In the mean time you are welcome to adjust the code:

https://github.com/PiInTheSky/pits/blob/master/tracker/aprs.c#L330

Cheers

rlharris, 2015-10-28:
But you did not follow those guidelines!  To wit:

             The APRS network is a shared resource and should not be 
abused. Due to the large
radio footprint of High Altitude Balloons (HAB) you should be 
conservative when you
set up your APRS tracker. Here are the recommended settings to use above 
5000 feet above the ground.
             Path: WIDE2-1is acceptable. No path is preferred.
             Transmit rate: 60 seconds or longer between transmissions.

DO NOT USE the following:
    Never use a 2 part path (WIDE1-1, WIDE2-1)
    Never use WIDE2-2 or WIDE3-3 in your path.
              Never transmit more than twice per minute

Upuaut, 2015-10-28:
Good point I think we misinterpreted it. Its in the list to fix thanks.

rlharris, 2015-10-28:
Something to consider (and I have guided the students at Profile School 
on how to do this) is add an "if then else" to change from no path above 
5000 feet (using the recommendation in the document) to WIDE2-1 at or 
below that level.

That would not cause undue traffic on the national frequency but would 
assist in recovery operations.

I can let you know how it works out after the students launch their 
balloon -- hopefully in November.

BTW, my compliments on nicely written code. While I developed firmware 
for embedded systems for almost 25 years, it has been over a decade 
since I even looked at any C source code. It was easy to find what I 
wanted to change and show the students how to make the change.

daveake, 2015-10-28:
I'm currently working on some AVR APRS code and will make similar changes
to the PITS code next week.

I've added settings for the altitude that the path switches at, and whether
above that altitude it uses no path or uses WIDE2-1; below that altitude it
uses WIDE1-1,WIDE2-1.

For PITS I'll probably allow for user-settable paths at altitude (WIDE2-1
or none) and at low altitude (anything you want).

Also I've added optional telemetry packets (e.g. for temperature, battery
voltage, number of satellites) and that code will make it into PITS soon.

Dave

rlharris, 2015-10-28:
Wicked cool on all points. Thanks for putting up with this grouchy old man.

On 10/28/2015 7:21 PM, David Akerman wrote:
> I'm currently working on some AVR APRS code and will make similar changes
> to the PITS code next week.
>
> I've added settings for the altitude that the path switches at, and
> whether
> above that [...]

-- 

Bob Harris
Can MOTCH Katmai Henry David Thoreau UDX Bda UD (1992-2005)
Longmeadow Bradford Torrey CD (2001-2014)
Ch Silvercreek Gossip Girl
Bath, NH

PiInTheSky, 2016-11-07:
Done (ages ago - should have been closed then).  Path is now configurable using APRS_HighPath (WIDE2-1 or none) for > APRS_Altitude, and WIDE1-1, WIDE2-1 below that.

closed 2016-11-07 by PiInTheSky