PiInTheSky/pits #12: Periodic Camera cant be enabled

laughingcrow, 2015-12-08:
I'm getting frequent ssh messages that the rpi camera is failing to enable:

>Running script take_pic_0 ...
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

So far, I've:
>sudo apt-get update
sudo apt-get upgrade
sudo rpi-update
sudo apt-get dist-upgrade

The camera is working and seems to be taking images, but isn't taking them successfully every time it should be. Has this been observed before, any suggestions as to a fix?

daveake, 2015-12-08:
Not heard of that, no.  As you probably know the camera code is external 
to the PITS code, so you're probably better off taking this to a 
raspberrypi.org forum.  All I can suggest is that you check the camera 
connections and the power supply / cable.  My guess would be that it's 
power-related.

Dave

laughingcrow, 2015-12-08:
Much appreciated, just thought I'd check to see if you guys had seen this
before!
Cheers,
Paddy

PiInTheSky, 2015-12-09:
Not a PITS issue.

closed 2015-12-09 by PiInTheSky