PiInTheSky/pits #14: Barometric Altitude.

ED6E0F17, 2015-12-25:
Atmospheric pressure is more accurate than GPS for calculating Ascent Rate at low altitudes. This should give a more stable/accurate landing prediction on the live tracker.

It would be interesting to compare GPS altitude at burst with the equivalent Ardupilot method:

https://github.com/diydrones/ardupilot/blob/master/libraries/AP_Baro/AP_Baro.cpp#L200

PiInTheSky, 2015-12-25:
Ardupilot's atmospheric model is inaccurate above 2km.  Readily available pressure sensors are not useful at 30km or above.  I have no intention of using barometric data to calculate the altitude.

closed 2015-12-25 by PiInTheSky

fab4space, 2015-12-25:
I hope you don't want to break an altitude record :p