PiInTheSky/pits #18: BME280 readings zero

Ryan-Ing, 2016-07-27:
I have set BME280=Y, and now I have the correct amount of fields but readings of zero. I have the VCC pin to 3Vo, SDA to SDI, SCL to SCK, Ground to ground. i2c is enabled. Do you know why this is?

daveake, 2016-07-28:
It should be enable_bme280=Y

Ryan-Ing, 2016-07-28:
It is set to that

daveake, 2016-07-28:
OK, not what you said.

Your BME280 isn't being seen on the i2c bus.  You can check what 
addresses are active using the i2cdetect program (find instructions for 
installation and use on the web).

The BME280 can be set to address ox76 or 0x77.  The software checks 
both.  With zero results, the device is not present on either, which 
means it's dead or not wired correctly.

Ryan-Ing, 2016-07-28:
The following addresses are active. Im pretty sure I have wired it correctly, is there any reason why it might be dead.

[image]

daveake, 2016-07-28:
> fullsizerender

Run the tracker program from the command line and see what it sees on 
startup.

Ryan-Ing, 2016-07-30:
It keeps showing "error writing to i2c slave" and it says the bme280 sensor is found at address 76h

daveake, 2016-07-30:
Do you have the option of setting yours to 76h ?

Ryan-Ing, 2016-07-30:
Im sorry, I know nothing about coding. So no, I dont think so

daveake, 2016-07-30:
No, I mean the device itself.  Does it have a jumper setting or similar to
change address?  This is a feature of the BME280.

Ryan-Ing, 2016-07-30:
Yes, I can add a jumper from SDO to GND, but it still says "error writing to i2c slave"

daveake, 2016-07-30:
Run i2detect with the jumper in each position and tell me what that says.

Ryan-Ing, 2016-07-30:
With the jumper SDO to GND: 
[image]

With it wired normally:

[image]

daveake, 2016-07-30:
Set to one address, close/restart the tracker program, see if it works.

If not, set to other address, close/restart the tracker program, see if it
works.

If it doesn't work in either setting then it's faulty.

The software autodetects which address it's on, and so should work with
either address.  I've only tested with one address as my BME280 board
doesn't have the option to change.  I don't recall which address mine uses
but it does work.

Ryan-Ing, 2016-07-30:
Aha! It's working now on the 76 address. Thank you very much for your help, I appreciate it!

closed 2016-07-30 by Ryan-Ing