PiInTheSky/pits #21: Number of telemetry packet get ignored after some time

zanco, 2016-11-08:
Hi, 
If I set the tracker to 1 telemetry packet for each 100 SSDV packets it works for the first images. However, within some hours after starting the transmission, I get 1 telemetry packet for 1 SSDV packet.

If I kill the tracker and start it again I get 1 telemetry for each 100 SSDV again, but within 24 hours I get 1 for 1 again.

Is there any log I need to add ?

Thanks,
Ben

daveake, 2016-11-08:
Latest software?  There were some changes 1-2 months ago to do with the
selection of telemetry vs images packets.

zanco, 2016-11-09:
Gitt pull says I am on the latest version, I update every time Daveake announces LoRa flights ;-)
. I do not recall to have seen this with  earlier versions. Restarted the tracker this eveneing and I am going to keep an eye on it.
Thanks,
Ben

daveake, 2016-11-09:
OK.  Let me know if it's LoRa or RTTY, and what image size please.

zanco, 2016-11-09:
Hi, it's LoRa, my images are usally 1280 * 720 and I am PE2BZ from the Netherlands

zanco, 2016-11-12:
After rebooting the Pi and running for 48 hours now, the problem did not return. All runs great at the moment.

daveake, 2016-11-12:
I checked here too, with larger images, and couldn't repeat the issue.

zanco, 2016-12-03:
No more problems in one month so I closed the issue

closed 2016-12-03 by zanco