PiInTheSky/pits #25: PITS will not turn on

sssendevour, 2017-04-18:
We have lights on our PI, and on the top parts of the stack but the PITS board will not turn on?
No lights on the warn and ok leds.

We have gone through the installation twice.

daveake, 2017-04-18:
1 - What model Pi ?
2 - Do the following:

login (ssh or keyboard/monitor)
cd pits/tracker
sudo killall tracker
sudo ./tracker

Then send a screenshot of what gets displayed when the tracker program
starts.

sssendevour, 2017-04-18:
That got it going. Not sure what happened but thank you for your super
quick reply.

I am a school teacher and two of my students have decided they are doing
this.

daveake, 2017-04-18:
Sounds like the Pi wasn't rebooted after installation - that starts the tracker program automatically.

Any more questions, please email support@pi-in-the-sky.com

closed 2017-04-18 by daveake