PiInTheSky/pits #26: Feature: Send command to Tracker to switch wifi on

xpix, 2017-08-01:
Hello, if it possible to send some commands to the tracker to trigger an action. I.e. here in germany it's possible that our payload is landed on the highest tree in the forest. A good solution on the RPI Zero W is to switch on an WIFI AP to get all the video and telemetry data. To save power, it will be very nice that i can send a command to the tracker that he switch his WIFI AP on, i can connect this via my phone and can download all relevant data.

If this possible and how can i implement this?

thx Frank

daveake, 2017-08-01:
I'm working on a generic upload facility which I hope to release later this
summer, and that will include the ability to run an external command, so
that would do what you need.

xpix, 2017-08-01:
Thanks for the fast answer! Wait on this Feature :)

daveake, 2021-04-20:
OK, it took 3 years, but uploads are now implemented as promised!

closed 2021-04-20 by daveake