PiInTheSky/pits #27: GPS ConnectionMode

jstjst, 2017-08-17:
I studied the gps.c code because I'm creating my own hardware. I saw the two ConnectionModes I2C and Serial/UART. So my question is it necessary to use both of these or can I just use UART.

Also, I ask me why you decide to bit-banging the I2C instead of using the normal one. My Suggestion is, that you want to avoid conflicts with other I2C components. Is this right? But is it also possible to use the normal one and use the second I2C bus for the other components?

daveake, 2017-08-18:
The software supports software I2C (used on the PITS board) or serial (used on some of my custom trackers).

PITS uses I2C because the serial port is used for the RTTY transmitter.

Software I2C is sued because the UBlox performs clock-stretching at points which are incompatible with the PI hardware I2C implementation (i.e. the latter is broken).

closed 2017-08-18 by daveake

jstjst, 2017-08-19:
Thanks for you fast answer