PiInTheSky/pits #29: Updating PITS software on Pi Zero

labjg, 2017-09-14:
Not really an issue, but a request..

I have a PITS-Zero board here, which was purchased earlier in the year and comes pre-assembled and with Raspbian Jessie Lite & PITS software pre-loaded onto the SD card.

I wanted to update PITS to get the latest features, so renamed the ~/pits directory and cloned and built the latest repo as per the PITS manual.  I also did an apt-get update and apt-get dist-upgrade.

Running the tracker in the terminal gave errors concerning wiringPi and PIGPIO, so re-installed those and cloned/built the PITS software again.  Things then seemed to work on reboot, but killing the startup tracker and instead running ./tracker in the terminal seems to stop images being sent (not clear why).  Also I noticed that the telemetry strings don't contain the internal temp data any more.

It would appear that I've upset something and might be best rolling back to my backup of the shipped SD card software, or starting from scratch with Stretch.

So really I'd just like to know how to properly update the PITS software in future.  I can't find any guidance on this.  Maybe an addition to the README would help others..?

I'd also welcome more explicit instructions on doing a fresh install for the Zero, as currently the manual just states that 'you don't need to do this because it's pre-installed'.  Is any of the Serial port and/or PiV3 bluetooth fussing applicable to the Zero?

Thanks folks.

labjg, 2017-09-14:
This is what I got when building the latest PITS code:

`pi@PiHAB1:~/pits $ ./build
Building tracker ...
gcc -Wall -O3  -o lora.o -c lora.c
gcc -Wall -O3  -o bme280.o -c bme280.c
gcc -Wall -O3  -o log.o -c log.c
gcc -Wall -O3  -o pipe.o -c pipe.c
gcc -Wall -O3  -o gps.o -c gps.c
gcc -Wall -O3  -o adc_i2c.o -c adc_i2c.c
gcc -Wall -O3  -o DS18B20.o -c DS18B20.c
gcc -Wall -O3  -o led.o -c led.c
gcc -Wall -O3  -o tracker.o -c tracker.c
gcc -Wall -O3  -o prediction.o -c prediction.c
gcc -Wall -O3  -o snapper.o -c snapper.c
gcc -Wall -O3  -o bmp085.o -c bmp085.c
gcc -Wall -O3  -o aprs.o -c aprs.c
gcc -Wall -O3  -o adc.o -c adc.c
gcc -Wall -O3  -o misc.o -c misc.c
gcc bme280.o misc.o bmp085.o led.o prediction.o pipe.o aprs.o tracker.o DS18B20.o lora.o log.o adc_i2c.o gps.o snapper.o adc.o  -lm -lwiringPi -lwiringPiDev -lpthread -lpigpio -lrt -o tracker
Copying configuration file ...
Enabling startup script.
Synchronizing state for tracker.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d tracker defaults
insserv: warning: script 'K01tracker' missing LSB tags and overrides
insserv: warning: script 'tracker' missing LSB tags and overrides
Executing /usr/sbin/update-rc.d tracker enable
update-rc.d: error: tracker Default-Start contains no runlevels, aborting.
Starting startup script
DONE
pi@PiHAB1:~/pits $`

labjg, 2017-09-14:
Update:  Did a fresh install of Stretch Lite and followed PITS manual to install everything else..

Same issues.  Startup process transmits image frames, but 'sudo ./tracker' does not.  No temperature measurement in telemetry string.

daveake, 2017-09-14:
sudo ./tracker just runs the tracker, and not the camera script.  If you view the startup script you'll see what it does.

Pi Zero (as supplied in the kits) does not have bluetooth; Pi Zero W does.

For updating see the section on that subject at the bottom of http://www.pi-in-the-sky.com/index.php?id=sd-card-image-from-scratch.  I'll get that added to the manual.

closed 2017-09-14 by daveake

labjg, 2017-09-14:
Ok, thanks.  Could have sworn I used to see things like 'sending image packet x of y' on the ./tracker terminal output.  Ho hum, no worries then.

Still not sure why the telemetry string internal temp measurement was showing on the old version but now is not.  I don't usually have GPS lock at my desk, but I can't imagine that's why.  Any ideas?

daveake, 2017-09-14:
It'll still show "sending image packet..." from the tracker if there's a
leftover image from before the camera script was stopped;l once that image
is sent then it'll revert to telemetry only.

Looks like the temperature code isn't being loaded if the software detects
it's running on a Zero; I'll get that sorted.  The temperature comes from
the GPU as there's no separate sensor on the PITS Zero board, but as the
code isn't being loaded then it's not measuring that either.

Dave

labjg, 2017-09-14:
Righto; thanks Dave.

emdeex, 2018-02-10:
I'm trying to update my Pi Zero from Uputronics as well.  It appears to have shipped (Jan 2018) with version "25/09/2016", which seems odd, considering the multiple updates since then.

The main problem I'm having is actually getting a terminal on the Pi Zero... I finally got a mini-HDMI convertor, but plugging it in to my display is getting a blank screen.

Is there is a SSH daemon running on the Pi by default, I can't see any details of it.  Getting a network adaptor running, considering not having a display working yet will be tricky.

Perhaps it would be a good idea to put a WiFi AP credentials section in the /boot/piesky.txt, so the tracker can connect to the local network and be accessible via ssh?

daveake, 2018-02-10:
We don't disable HDMI output by default, so that should work as on any Pi.
Be sure to have the monitor connected before you apply power.

ssh should be enabled on the supplied SD card, so if you connect a USB LAN
adapter can get the IP address from your router and connect that way.

Or you can put your WiFi details on the card.  I've not done yet myself but
it's standard Raspbian (and nothing to do with pisky.txt).

The supplied pits software is fully functional and there have been no
significant changes since then for pits zero.

Dave

emdeex, 2018-02-10:
thanks Dave, not sure why the display is not working... I'll keep trying.

I'll try the USB LAN adapter... what's the credentials required for SSH?

With the WiFi details, any hints on getting that working without being able to log in yet?  Are there settings on the /boot/ directory I can tweak?

daveake, 2018-02-10:
We've not changed the normal pi/raspberry login.

I don't know how the WiFi stuff is done as I've never set it up that way.

Dave