PiInTheSky/pits #34: PITS Zero; interface and wifi

chessnecessary, 2019-06-18:
Hello, currently I am working with my PITS Zero (2 of them) to try and transmit photos from our GoPros to our LoRa gateway. 
So far, I have successfully been able to transmit images from a USB webcam connected to the PITS Zero to the LoRa gateway and have them show up on HabHub's tracker site. I have also successfully sent wifi commands to the GoPro via the LoRa Gateway. This was easier to do, as that raspberry pi gives me access to a windows-esque interface.

My current issue is two-fold: I cannot seem to determine if I can activate a similar interface on the PITS Zero (currently I only have access to a full-screen terminal). Secondly, and partly due to the first, I cannot seem to obtain an internet connection via wifi in order to download any of the software/packages needed for implementation. I'm fairly sure this version of the PITS Zero has wifi capability, and I have attempted to connect to wifi using only the Linux terminal--but to no avail so far.

Any advice/help/etc. anyone is able to give would be greatly appreciated!

chessnecessary, 2019-06-20:
Is this github still monitored at all? Any help would be appreciated!

daveake, 2019-06-20:
I replied by email 2 days ago.

To connect to a GoPro by WiFi you need the W version of the PiZero.  To also have internet access you will need to connect to a LAN using a USB-LAN adapter.

I'm closing this issue as these are generic Pi Zero questions and not to do with the PITS source code.  You are welcome to ask questions like this to the PITS support email address, or just ask in the Raspberry Pi forums.

closed 2019-06-20 by daveake

chessnecessary, 2019-06-24:
David,

Not sure if you will receive this, as it is in response to the
"notifications" email at github, but nonetheless:

I did not receive your initial email, if you sent one, which is what
prompted my second posting to the PITS github issue page. I apologize for
bringing up an irrelevant issue but was having trouble executing many of
the methods for wifi connection I had found from other sources. This caused
me to assume that something relevant to the PITS source code/boot files was
negating the methods I was attempting--so I thought that asking you and the
community would be appropriate.

If you don't mind, could you try to send the first email again (if it had
any helpful advice)? Thank you!

Chess

daveake, 2019-06-24:
No there's nothing about the pits installation that affects the WiFi iin
any way.

The easiest way of getting a zero online is to purchase a usb-lan adapter
and plug that in to a wider network.  Generally no configuration is needed
and it works straight away.

For wireless I just add the SSID, password etc to the text file
/etc/wpa_supplicant/wpa_supplicant.conf.  Works for me.

chessnecessary, 2019-06-24:
David,

Alright, I will continue trying then! Thank you for your responses.