PiInTheSky/pits #36: Required information on pisky file

didaquis, 2019-09-23:
Hi!

Information about parameters and values in pisky.txt file are required.
For exmample, what is the meaning of "LORA_Mode" setting? 

Please, add detailed instrucctions on README.md file

daveake, 2019-09-23:
See the manual  http://www.pi-in-the-sky.com/data/uploads/pits-manual.pdf

didaquis, 2019-09-23:
Thanks for that link.

In my opinion documentation of PiInTheSky kit i's not friendly. In the way the documentation is written, it can only be understood by those who already understand all the concepts and know how to assemble and configure the kit.

Will be cool add some kind of baby steps in a lot of concepts, for example to build antennas.

closed 2019-09-23 by didaquis

daveake, 2019-09-23:
Your experience is noted but is the opposite of what we have been told 
by others.

High altitude ballooning is a wide-ranging subject and is not for 
everyone.  We refer all purchasers to the introductory guide (which I 
also wrote) here - http://www.daveakerman.com/?p=1732. You mentioned 
assembly, which is shown step-by-step in the manual.  You mention 
configuration which, again, is covered in the manual.

For the antenna, the manual seems to be missing a link to this video 
showing how to make one: https://www.youtube.com/watch?v=xemKyV1H6Xs

Dave