PiInTheSky/pits #38: PITS internal temperature not updating on pi zero

MGPZ, 2020-02-28:
Hi

When using the PITS tracker on a Pi Zero (without the PITSZero board, just GPS and LoRa modules attached) the internal temperature reading is not updating so the initial reading taken from the GPU just repeats for the duration it is running. 

If I restart the tracker, an updated temperature reading is reported but again this repeats without updating.

Many Thanks

Mark

PiInTheSky, 2020-02-28:
Thanks, sounds like it's only checking once or is re-reading the same data
for some reason.  I'll check over the weekend.

MGPZ, 2020-02-28:
Happy to help.

Mark

PiInTheSky, 2020-03-02:
Fixed in latest issue.

It was setting a flag that said it had found a DS18B20, and that stopped further checks of the GPU temperature.

closed 2020-03-02 by PiInTheSky