PiInTheSky/pits #39: Image resend after 255 images

KevWal, 2021-02-21:
Hi

Thanks very much for the great code.  I do seem to have an issue getting pits to resend image packets after the 255th image though.  I assume due to the wrap of image numbers that happens on the ground, vs not wrapping the image numbers in pits?  So I get a situation like this:

`LORA0: $$KW03PL,1531,12:32:56,52.32331,-0.70613,00034,0,14,11,26.2,4.860,0.66,0.00000,0.00000,0.0,0,-49,6,49,N*28B1
LORA0: SSDV image 682 packet 40 of 130
LORA0: SSDV image 682 packet 41 of 130

Uplink period ...
Setting frequency to 434.735MHz for uplink
Channel 0 frequency 434.735000 FrequencyValue = 6CAF0Ah
Set Uplink Mode to 6
Listening on LoRa channel 0
Rx LoRa channel 0
*** Received 255 bytes
LORA0: Frequency Difference =  0.1kHz
SSDV uplink message !169:130=53,96,170:34=20,33
Uplink: !169:130=53,96,170:34=20,33

MarkMissingPacketsBeyond(2, 169, 130)
FindImageInList - NOT FOUND
MarkMissingPackets(2, 169, 53, 53)
FindImageInList - NOT FOUND
MarkMissingPackets(2, 169, 96, 96)
FindImageInList - NOT FOUND
MarkMissingPacketsBeyond(2, 170, 34)
FindImageInList - NOT FOUND
MarkMissingPackets(2, 170, 20, 20)
FindImageInList - NOT FOUND
MarkMissingPackets(2, 170, 33, 33)
FindImageInList - NOT FOUND

Channel 0 frequency 434.735000 FrequencyValue = 6CAF0Ah
Reset after Uplink Mode

LORA0: SSDV image 682 packet 42 of 130
LORA0: SSDV image 682 packet 43 of 130`

Where as after a reset it works fine:

`LORA0: $$KW03PL,7,12:45:53,52.32283,-0.70614,00115,0,193,10,34.9,4.836,0.66,0.00000,0.00000,0.0,0,-49,6,1,N*9C46
LORA0: SSDV image 2 packet 32 of 80
LORA0: SSDV image 2 packet 33 of 80

Uplink period ...
Setting frequency to 434.735MHz for uplink
Channel 0 frequency 434.735000 FrequencyValue = 6CAF0Ah
Set Uplink Mode to 6
Listening on LoRa channel 0
Rx LoRa channel 0
*** Received 255 bytes
LORA0: Frequency Difference =  0.2kHz
SSDV uplink message !2:26=0-2
Uplink: !2:26=0-2

MarkMissingPacketsBeyond(2, 2, 26)
MarkMissingPackets(2, 2, 0, 2)
Index = 0
Channel 0 frequency 434.735000 FrequencyValue = 6CAF0Ah
Reset after Uplink Mode

LORA0: SSDV image 2 packet 1 of 80 ** RESEND **
LORA0: SSDV image 2 packet 2 of 80 ** RESEND **
LORA0: SSDV image 2 packet 3 of 80 ** RESEND **
LORA0: SSDV image 2 packet 34 of 80
LORA0: SSDV image 2 packet 35 of 80`

Works close to 255:
`LORA0: SSDV image 238 packet 3 of 48
LORA0: SSDV image 238 packet 4 of 48
Uplink period ...
Setting frequency to 434.737MHz for uplink
Channel 0 frequency 434.737000 FrequencyValue = 6CAF2Bh
Set Uplink Mode to 6
Listening on LoRa channel 0
Rx LoRa channel 0
*** Received 255 bytes
LORA0: Frequency Difference = -0.1kHz
SSDV uplink message !237:46=32-33
Uplink: !237:46=32-33

MarkMissingPacketsBeyond(2, 237, 46)
MarkMissingPackets(2, 237, 32, 33)
Index = 1
Channel 0 frequency 434.737000 FrequencyValue = 6CAF2Bh
Reset after Uplink Mode
>>>> Switching to SSDV file ssdv_2_237.bin
LORA0: SSDV image 237 packet 33 of 49 ** RESEND **
LORA0: SSDV image 237 packet 34 of 49 ** RESEND **
>>>> Switching to SSDV file ssdv_2_238.bin
LORA0: SSDV image 238 packet 5 of 48
LORA0: SSDV image 238 packet 6 of 48
LORA0: $$KW03PL,366,17:23:11,52.32273,-0.70619,00132,0,117,13,13.4,4.810,0.66,0.00000,0.00000,0.0,0,-48,6,81,N*451D
`

But then just past 255:

`LORA0: $$KW03PL,478,17:53:54,52.32272,-0.70618,00136,0,117,18,13.9,4.837,0.66,0.00000,0.00000,0.0,0,-49,6,111,N*9E1E
LORA0: SSDV image 263 packet 28 of 50
LORA0: SSDV image 263 packet 29 of 50
LORA0: SSDV image 263 packet 30 of 50
Uplink period ...
Setting frequency to 434.737MHz for uplink
Channel 0 frequency 434.737000 FrequencyValue = 6CAF2Bh
Set Uplink Mode to 6
Listening on LoRa channel 0
Battery Voltage = 4.852799
Rx LoRa channel 0
*** Received 255 bytes
LORA0: Frequency Difference = -0.0kHz
SSDV uplink message !7:22=14
Uplink: !7:22=14

MarkMissingPacketsBeyond(2, 7, 22)
FindImageInList - NOT FOUND
MarkMissingPackets(2, 7, 14, 14)
FindImageInList - NOT FOUND
Channel 0 frequency 434.737000 FrequencyValue = 6CAF2Bh
Reset after Uplink Mode
LORA0: SSDV image 263 packet 31 of 50
LORA0: SSDV image 263 packet 32 of 50`

Thanks very much
Kevin

daveake, 2021-02-21:
Ta I'll have a look.  Not something that would happen on a typical flight
which wouldn't get to that high an image number.

daveake, 2021-04-20:
Fixed in version being released this week.

closed 2021-04-20 by daveake