PiInTheSky/pits #40: PITS does not listen for resend uplink when GPS -> Sats not > 0

KevWal, 2021-02-27:
With the potential GPS issues caused by the Pi I have been doing some testing of that scenario, and one thing that breaks is SSDV Resend Uplink - because the code checks for GPS -> Sats > 0.

I have tested replacing this with simply GPS -> SecondsInDay > 0, to make sure it has a time, so it knows when to listen, but does not need an active GPS signal.

I was thinking about doing a Pull request to make that change - is that something you would support?

Thanks very much, Kevin

PiInTheSky, 2021-02-27:
Well, the reason for that is that it needs an accurate clock, and the
tracker will only have an accurate clock if it's had a valid GPS time.  It
could just test whether it's had valid GPS time at some point rather than
currently having one.

But, just fix the GPS :)

KevWal, 2021-02-27:
Id love to just fix the GPS - but that means a £20 active aerial, which double the cost of the payload compared to my current 1/4 whip - which to be fair has actually done fine on this flight. :)

Given how accurately the GPS clock maintains time once it has the time, I have tested:

+       // KW What if we loose GPS in flight?  We still keep time.
+       // printf("KW GPS->SecondsInDay %ld\n", GPS->SecondsInDay);
+       // if (GPS->Satellites > 0)
+       if (GPS->SecondsInDay > 0)

in TDMTimeToSendOnThisChannel and UplinkTimeToSendOnThisChannel.

Just my 2p, up to you obviously :)

Cheers
Kev

daveake, 2021-02-27:
Yeah it's a simple change, so I'll do it.

daveake, 2021-04-20:
Done in version being issued this week.

closed 2021-04-20 by daveake