PiInTheSky/pits #43: Small issue with FrequencyError Calculation

KevWal, 2021-03-05:
The calculation of FrequencyError in Lora.c seems to be done based on TX bandwidth currently, rather than RX bandwidth.  My code used to test, and then fix the issue below from Lora.c in double FrequencyError(int Channel):

```
        printf("FrequencyError Now using BandwidthInKHz = %d.\n", LoRaModes[Config.LoRaDevices[Channel].UplinkMode].Bandwidth);
        printf("FrequencyError Was using BandwidthInKHz = %f.\n", BandwidthInKHz(Channel));
        // return - ((double)Temp * (1<<24) / 32000000.0) * (BandwidthInKHz(Channel) / 500.0); // Was dividing by tx bandwidth not rx bandwidth?
        return - ((double)Temp * (1<<24) / 32000000.0) * (LoRaModes[Config.LoRaDevices[Channel].UplinkMode].Bandwidth / 500.0);
```

PiInTheSky, 2021-03-05:
Thanks, will check this soon.

Dave

daveake, 2021-04-20:
Incorporated into latest version going out this week, thanks!

closed 2021-04-20 by daveake