PiInTheSky/pits #45: ADC present on custom board but can't be used.

KevWal, 2022-07-31:
Currently the code only enables the ADC if the BoardType is designed to have the ADC and Config.DisableADC is False.

This gives no option to enable the ADC for boards that by default don't have it, but it has been added.

Editing misc.c to read:

```
        // Battery voltage and current, if available
        if (((Config.BoardType == 3) || (Config.BoardType == 4) || (Config.DisableADC)) & (!Config.EnableADCVolt))
        {
                        // Pi Zero - no ADC on the PITS Zero, or manually disabled ADC
        }
        else if ((Config.BoardType == 0) || (Config.EnableADCVolt))
        {
                // Pi A or B.  Only Battery Voltage on the PITS

                sprintf(ExtraFields1, ",%.3f", GPS->BatteryVoltage);
                if (ShowFields) printf(",Volts");
        }
        else
        {
                // Pi A+ or B+ (V1 or V2 or V3).  Full ADC for voltage and current

                sprintf(ExtraFields1, ",%.1f,%.3f", GPS->BatteryVoltage, GPS->BoardCurrent);
                if (ShowFields) printf(",Volts,Current");
        }
```

and tracker.c to create the thread:

```
        // DisableADC is used to disable the ADC on boards that should have an ADC (but dont)
        // EnableADCVolt is used to enable the ADC on boards that should not have an ADC (but do)
        if (((Config.BoardType != 3) && (Config.BoardType != 4) && (!Config.DisableADC)) || (Config.EnableADCVolt))
        {
                // Not a zero, so should have ADC on it
                if (I2CADCExists())
                {
                        printf ("V2.4 or later board with I2C ADC\n");

                        if (pthread_create(&ADCThread, NULL, I2CADCLoop, &GPS))
                        {
                                fprintf(stderr, "Error creating ADC thread\n");
                                return 1;
                        }
                }
                else
                {
                        printf ("Older board with SPI ADC\n");

                        if (Config.LoRaDevices[0].InUse)
                        {
                                printf ("Disabling SPI ADC code as LoRa CE0 is enabled!!\n");
                                Config.DisableADC = 1;
                        }
                        else
                        {
                                if (pthread_create(&ADCThread, NULL, ADCLoop, &GPS))
                                {
                                        fprintf(stderr, "Error creating ADC thread\n");
                                        return 1;
                                }
                        }
                }
        }
```

and tracker.c to read the config:

```
        // DisableADC is used to disable the ADC on boards that should have an ADC (but dont)
        // EnableADCVolt is used to enable the ADC on boards that should not have an ADC (but do)
        ReadBoolean(fp, "Disable_ADC", -1, 0, &(Config->DisableADC));
        ReadBoolean(fp, "Enable_ADC_Volt", -1, 0, &(Config->EnableADCVolt));
```

and finally misc.h:

```
        int DisableADC;
        int EnableADCVolt;
```

Allows the ADC to be used to read voltage on a board that does not usually have the ADC present.

Happy to sort this as a pull request if easier.

Thanks very much
Kevin

daveake, 2022-07-31:
Please do a pull request - I'm away on site for a couple of days so I don't
have any time to edit it myself.  I'll approve once it's done.

KevWal, 2022-08-04:
Hi, Pull request created - https://github.com/PiInTheSky/pits/pull/46 - any issues do let me know, no rush of course.

KevWal, 2022-08-20:
Apologises Dave, I messed up the pull request and included the whole of my develop branch.  Could you undo urgently, and I will redo the pull request.  Sorry :(

PiInTheSky, 2022-08-21:
Reverted.

closed 2022-08-21 by PiInTheSky

KevWal, 2022-08-21:
Thanks Dave, I know what I did wrong, I made a branch with just the correct commits, submitted the pull request, but then carried on making commits on the same branch!  

I will make a feature branch dedicated to the changes and resubmit 😀