PiInTheSky/pits #49: Landing predictions become inf / nan

KevWal, 2022-08-23:
Playing back this GPS track:  https://github.com/KevWal/pits/blob/63f0a1918b78d8278c807f693a08f81b01c122c2/nmea/Friday-No-GSV-VTG.NMEA

I get issues with the prediction code:

> LORA0: $$KW01,236,21:03:57,52.01939,-0.15447,17022,35,83,5,31.3,4.317,0.99,52.29643,2.62463,3.3,3333,-70,6,22,N*BA6C
> ...
> LORA0: $$KW01,237,21:04:16,52.01968,-0.15169,16838,1,0,5,31.2,4.273,0.97,52.29172,2.58097,3.3,3270,-64,8,23,N*8479
> ...
> LORA0: $$KW01,238,21:04:27,52.01996,-0.14920,16674,35,83,5,31.2,4.356,inf,    nan,    nan,0.0,2147483647,-64,8,23,N*1051
> ...
> LORA0: $$KW01,239,21:04:27,52.02020,-0.14700,16529,35,77,5,31.2,4.311,inf,    nan,    nan,0.0,2147483647,-64,8,23,N*BCD6

Playing it back a second time, I get the same thing (Scroll back buffer didnt go back far enough to find the start of the issue):

> LORA0: $$KW01,254,22:39:14,52.03025,0.02985,11048,87,91,5,30.4,4.350,inf,    nan,    nan,0.0,2147483647,-62,6,64,N*6B11
> ...
> LORA0: $$KW01,255,22:39:30,52.03013,0.03556,10942,87,91,5,30.6,4.276,inf,    nan,    nan,0.0,2147483647,-62,6,64,N*0E80
> ...
> LORA0: $$KW01,256,22:39:47,52.02990,0.04163,10833,87,92,5,30.6,4.353,inf,    nan,    nan,0.0,2147483647,-62,6,64,N*3319

Interestingly, 2 values are fixed again at the very end of the playback:

> LORA0: SSDV image 176 packet 79 of 114
> LORA0: $$KW01,408,23:19:34,51.99414,0.58410,00126,20,92,5,31.0,4.346,inf,    nan,    nan,0.0,2147483647,-60,6,95,N*C940
> LORA0: SSDV image 176 packet 80 of 114
> LORA0: $$KW01,409,23:19:35,51.99412,0.58437,00116,20,92,5,31.0,4.263,inf,    nan,    nan,0.0,2147483647,-60,6,95,N*2BA4
> LORA0: SSDV image 176 packet 81 of 114
> LORA0: $$KW01,410,23:19:35,51.99412,0.58454,00109,20,92,5,31.0,4.263,inf,    nan,    nan,0.0,2147483647,-60,6,95,N*D6F9
> LORA0: ^^KW01,434.712,1,2,48,96,0*07DE
> LORA0: SSDV image 176 packet 82 of 114
> LORA0: $$KW01,411,23:19:45,51.99410,0.58506,00090,14,90,5,30.9,4.263,inf,51.99411,0.58481,0.0,0,-60,6,95,N*AA24
> LORA0: SSDV image 176 packet 83 of 114
> LORA0: $$KW01,412,23:19:47,51.99410,0.58519,00084,1,0,5,30.9,4.300,inf,51.99410,0.58519,0.0,0,-60,6,95,N*1E58
> LORA0: SSDV image 176 packet 84 of 114
> LORA0: $$KW01,413,23:19:53,51.99410,0.58560,00065,14,90,5,30.9,4.300,inf,51.99410,0.58519,0.0,0,-60,6,95,N*FFFA

Cheers
Kev

PiInTheSky, 2022-08-24:
When you say "Playing it back a second time", are you replacing the GPS
file while the tracker program is running, or is this a separate run of the
tracker?

PiInTheSky, 2022-08-24:
Can you send me your NMEA file please, and I'll set this up today.

Dave

KevWal, 2022-08-24:
Hi, the 'second run' was a totally independent run of the tracker code.  The tracker code exits at the end of the file, and I re run the program from the command line.

Fyi, I had the same issue on a different set of test data last night, I will re run that today and capture it in a new comment here.

Thanks
Kev

KevWal, 2022-08-24:
> Can you send me your NMEA file please, and I'll set this up today. Dave

Hi Dave,

File attached (I had to add .txt at the end to let me upload it), or also linked on gihub in the mesage above.

Cheers
Kev

[Friday-No-GSV-VTG.NMEA.txt](https://github.com/PiInTheSky/pits/files/9414756/Friday-No-GSV-VTG.NMEA.txt)

KevWal, 2022-08-24:
Second example (https://github.com/KevWal/pits/blob/develop/nmea/XABEN11.NMEA) re running, currently it is on the way up, it broke on the way down here last night:  

[image]

(where the descent becomes vertical)

KevWal, 2022-08-24:
Just caught the break before it scrolled out of the buffer! :)

> LORA0: $$KW01,690,12:43:13,52.06396,0.65472,24148,3,0,5,28.1,4.356,1.29,51.96360,1.03043,2.9,4435,0,0,0,N*3E5E
> LORA0: $$KW01,691,12:43:38,52.06362,0.65289,23761,0,0,5,28.0,4.305,1.75,51.94694,1.09340,2.5,5149,0,0,0,N*ACC0
> LORA0: $$KW01,692,12:43:54,52.06353,0.65208,23532,0,0,5,28.1,4.199,1.67,51.94973,1.08321,2.5,5016,0,0,0,N*6A25
> LORA0: $$KW01,693,12:44:18,52.06343,0.65094,23189,22,265,5,28.3,4.366,inf,    nan,    nan,0.0,2147483647,0,0,0,N*B52C
> LORA0: $$KW01,695,12:44:51,52.06330,0.64941,22732,22,265,5,28.4,4.252,inf,    nan,    nan,0.0,2147483647,0,0,0,N*8601
> LORA0: $$KW01,696,12:45:11,52.06321,0.64857,22427,0,0,5,28.3,4.363,inf,    nan,    nan,0.0,2147483647,0,0,0,N*4276

Will leave it running as yesterday it fixed its self at the very end.

Cheers
Kev

PiInTheSky, 2022-08-24:
The inf is CDA which implies zero descent rate, so the issue will be
somewhere around the use of altitude.

I ran my NMEA twice, no issues.  Currently running XABEN11.

Dave

KevWal, 2022-08-24:
Want to send me your NMEA and I can run here once https://github.com/KevWal/pits/blob/develop/nmea/XABEN11.NMEA is finished?

KevWal, 2022-08-24:
XABEN11 now finished here, and as per my run last night it worked at the very end:

> LORA0: $$KW01,927,13:47:06,51.98930,0.95826,00109,0,0,5,31.0,4.246,inf,    nan,    nan,0.0,2147483647,0,0,0,N*E15C
> LORA0: $$KW01,928,13:47:09,51.98926,0.95841,00103,39,103,5,31.0,4.246,inf,    nan,    nan,0.0,2147483647,0,0,0,N*AE8B
> LORA0: $$KW01,929,13:47:11,51.98918,0.95871,00090,0,0,5,31.1,4.246,inf,51.98923,0.95856,0.0,0,0,0,0,N*9898
> LORA0: $$KW01,930,13:47:14,51.98915,0.95886,00083,0,0,5,31.1,4.246,inf,51.98923,0.95856,0.0,0,0,0,0,N*466D

Appretiate yuo looking into this.

Cheers
Kev

daveake, 2022-08-24:
> Inf is expected at the end, when the altitude stops reducing.  It's not
> unusual for this to happen before the code recognises the landing and
> disables the prediction code.
>
> Any inf value prior to landing is wrong.
>
> Dave

Still no inf or nan.  I'll run XABEN11 again ...

Can you send your pisky.txt please.

Dave

KevWal, 2022-08-25:
Sure - https://github.com/KevWal/pits/blob/develop/boot/pisky.txt

daveake, 2022-08-25:
Thanks.  Second run with XABEN11 showed no inf or nan.  I'll try with your
settings next.

KevWal, 2022-08-25:
This morning I turned all your printf's on inside prediction.c and now I can no longer reproduce it either!  

I might just add some 'isnormal()' guards around those values for now so it doesn't break the sentance and keep an eye on it..

Cheers
Kev

daveake, 2022-08-25:
Try just printing the values used in the CDA calculation and see what you
get.

KevWal, 2022-08-25:
And I now no longer get it to fail, tried both NMEA files, tried printing or not printing debug stuff, just works every time now, and I cant think of anythnig changed!

I have now added some isnormal()  code just in case, dont want it breaking again in flight...

```
        // Landing Prediction, if enabled and not a seperate sentence
        if (Config.EnableLandingPrediction && (Config.PredictionID[0] == '\0'))
        {
                if (!isnormal(GPS->CDA))
                {
                        printf("ERROR: GPS->CDA not normal!\n");
                        GPS->CDA = 0;
                }
                if (!isnormal(GPS->PredictedLatitude))
                {
                        printf("ERROR: GPS->PredictedLatitude not normal!\n");
                        GPS->PredictedLatitude = 0;
                }
                if (!isnormal(GPS->PredictedLongitude))
                {
                        printf("ERROR: GPS->PredictedLongitude not normal!\n");
                        GPS->PredictedLongitude = 0;
                }
                if (!isnormal(GPS->PredictedLandingSpeed))
                {
                        printf("ERROR: GPS->PredictedLandingSpeed\n");
                        GPS->PredictedLandingSpeed = 0;
                }
```

Thanks for trying to reproduce.  I'll come back here if I see it again!

Cheers
Kev

PiInTheSky, 2022-08-25:
OK, I'll protect the CDA code so if there new CDA is silly then it uses the
existing value.

It's based on altitude changes, and is slugged to reduce the effect of GPS
noise and vertical wind gusts.  Might be an error around there.

Dave

daveake, 2022-10-11:
Inf is expected at the end, when the altitude stops reducing.  It's not
unusual for this to happen before the code recognises the landing and
disables the prediction code.

Any inf value prior to landing is wrong.

Dave